TobKra96/music_led_strip_control #163: Effects don't work properly after RPi4 Restart

VisualHakuna, 2022-06-01:
When my RPi4 Boots up. MLSC starts but doesn't run the effects correctly. To fix this I have to go to device settings and click save every time I reboot my RPi. I have 59*2LED's without an inverter. Running off of the 5V Pin2 and the data line from GPIO18. Everything seems to be set correctly with the defaults. 

When the effects are not running correctly; all but the first 3 LED's are frozen. The first 3 are doing seemingly random actions.
Again, going to device settings and just clicking save fixes it.
Any suggestions?

VisualHakuna, 2022-06-02:
https://pastebin.com/GqFWuuVU - config_service.py
https://pastebin.com/gpyxTRr3 - config.json

Not entirely sure which config you meant. I will try and figure out how to run dev_2.3

VisualHakuna, 2022-06-03:
Reinstalled Raspberian OS. Reinstalled MLSC and PROPERLY read limitations in the installation guide. Blacklisted the snd_bcm2835

https://github.com/TobKra96/music_led_strip_control/wiki/Installation-Guide#i-local-raspberry-pi-for-output

closed 2022-06-03 by VisualHakuna

marcelgerber, 2023-04-08:
Just noting here that I ran into the exact same problem (mlsc was behaving weirdly when the service was started directly after boot, but was behaving normally if I restarted it or saved the device).
For me, the solution was also to blacklist `snd_bcm2835`.