TobKra96/music_led_strip_control #166: Music effects not working

Xitee1, 2022-06-06:
I have installed MLSC on my Raspberry Pi with this https://github.com/nicokaiser/rpi-audio-receiver programm so I can send the data with Bluetooth to the RPi. I have selected the "microphone" (BL audio).
I have a ESP8266 with WLED connected with MLSC which is working just fine with the Non-Music effects.
However if I want a Music-Effect the connection to WLED times out (I think - the light pauses and stays on for a few seconds until it goes off and WLED is no longer saying that it is used through the API).
Can anyone help me?

[image]

TobKra96, 2022-06-11:
Hi,

looks like an issue with pyaudio and your music/input source. Does this error occurs, if you use the default device?

FloKTC, 2023-06-06:
i have the same problem ( with the pi3) i try to config a virtual mic at the moment  , unfortunately unsuccessful so far

TobKra96, 2023-06-06:
Yes, virtual mics are hard to config and buggy at the moment. At the moment I could not find a solution for virtual mics.

closed 2023-06-06 by TobKra96