TobKra96/music_led_strip_control #168: something is wrong

Marshalldog, 2022-06-10:
```
pi@raspberrypi:~ $ sudo apt-get update
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [44                                                                              .1 kB]
Hit:2 http://deb.debian.org/debian bullseye InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:4 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main armhf Pa                                                                              ckages [154 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 Pa                                                                              ckages [152 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main Translat                                                                              ion-en [95.3 kB]
Get:8 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [2,600 B                                                                              ]
Get:9 http://deb.debian.org/debian bullseye-updates/main armhf Packages [2,604 B                                                                              ]
Get:10 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [279 k                                                                              B]
Get:11 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [273 k                                                                              B]
Fetched 1,065 kB in 4s (296 kB/s)
Reading package lists... Done
pi@raspberrypi:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  cifs-utils dphys-swapfile dpkg dpkg-dev gzip libcamera0 libdpkg-perl
  libldap-2.4-2 libldap-common liblzma5 libpam-chksshpwd libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin
  libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libssl1.1 libxml2
  openssl pi-bluetooth raspi-config rpi-eeprom rsyslog userconf-pi xz-utils
28 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.9 MB of archives.
After this operation, 14.3 MB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:1 http://security.debian.org/debian-security bullseye-security/main arm64 dp                                                                              kg arm64 1.20.10 [2,513 kB]
Get:2 http://security.debian.org/debian-security bullseye-security/main arm64 gz                                                                              ip arm64 1.10-4+deb11u1 [129 kB]
Get:3 http://security.debian.org/debian-security bullseye-security/main arm64 rs                                                                              yslog arm64 8.2102.0-2+deb11u1 [681 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 li                                                                              blzma5 arm64 5.2.5-2.1~deb11u1 [165 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main arm64 xz                                                                              -utils arm64 5.2.5-2.1~deb11u1 [219 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 ci                                                                              fs-utils arm64 2:6.11-3.1+deb11u1 [87.8 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main arm64 dp                                                                              kg-dev all 1.20.10 [2,306 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam0g arm64 1                                                                              .4.0-9+deb11u1+rpt2 [128 kB]
Get:9 http://security.debian.org/debian-security bullseye-security/main arm64 li                                                                              bdpkg-perl all 1.20.10 [1,545 kB]
Get:10 http://security.debian.org/debian-security bullseye-security/main arm64 l                                                                              ibldap-2.4-2 arm64 2.4.57+dfsg-3+deb11u1 [222 kB]
Get:11 http://security.debian.org/debian-security bullseye-security/main arm64 l                                                                              ibldap-common all 2.4.57+dfsg-3+deb11u1 [95.8 kB]
Get:12 http://security.debian.org/debian-security bullseye-security/main arm64 l                                                                              ibxml2 arm64 2.9.10+dfsg-6.7+deb11u2 [629 kB]
Get:13 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam-modules-                                                                              bin arm64 1.4.0-9+deb11u1+rpt2 [107 kB]
Get:14 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam-modules                                                                               arm64 1.4.0-9+deb11u1+rpt2 [318 kB]
Get:15 http://archive.raspberrypi.org/debian bullseye/main arm64 dphys-swapfile                                                                               all 20100506-7+rpt1 [22.0 kB]
Get:16 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam-runtime                                                                               all 1.4.0-9+deb11u1+rpt2 [208 kB]
Get:17 http://archive.raspberrypi.org/debian bullseye/main arm64 libssl1.1 arm64                                                                               1.1.1n-0+deb11u2+rpt1 [1,391 kB]
Get:18 http://archive.raspberrypi.org/debian bullseye/main arm64 libcamera0 arm6                                                                              4 0~git20220510+22656360-1 [764 kB]
Get:19 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam-chksshpw                                                                              d arm64 1.4.0-9+deb11u1+rpt2 [84.3 kB]
Get:20 http://archive.raspberrypi.org/debian bullseye/main arm64 libraspberrypi-                                                                              dev arm64 1:2+git20220324~090146+c4fd1b8-1 [129 kB]
Get:21 http://archive.raspberrypi.org/debian bullseye/main arm64 libraspberrypi-                                                                              bin arm64 1:2+git20220324~090146+c4fd1b8-1 [70.1 kB]
Get:22 http://archive.raspberrypi.org/debian bullseye/main arm64 libraspberrypi0                                                                               arm64 1:2+git20220324~090146+c4fd1b8-1 [78.3 kB]
Get:23 http://archive.raspberrypi.org/debian bullseye/main arm64 libraspberrypi-                                                                              doc all 1:2+git20220324~090146+c4fd1b8-1 [1,908 B]
Get:24 http://archive.raspberrypi.org/debian bullseye/main arm64 openssl arm64 1                                                                              .1.1n-0+deb11u2+rpt1 [831 kB]
Get:25 http://archive.raspberrypi.org/debian bullseye/main arm64 pi-bluetooth al                                                                              l 0.1.19 [5,764 B]
Get:26 http://archive.raspberrypi.org/debian bullseye/main arm64 raspi-config al                                                                              l 20220506 [30.4 kB]
Get:27 http://archive.raspberrypi.org/debian bullseye/main arm64 rpi-eeprom arm6                                                                              4 13.14-1 [1,133 kB]
Get:28 http://archive.raspberrypi.org/debian bullseye/main arm64 userconf-pi all                                                                               0.2 [5,736 B]
Fetched 13.9 MB in 5s (2,642 kB/s)
Reading changelogs... Done
Preconfiguring packages ...
(Reading database ... 35576 files and directories currently installed.)
Preparing to unpack .../dpkg_1.20.10_arm64.deb ...
Unpacking dpkg (1.20.10) over (1.20.9) ...
Setting up dpkg (1.20.10) ...
(Reading database ... 35585 files and directories currently installed.)
Preparing to unpack .../gzip_1.10-4+deb11u1_arm64.deb ...
Unpacking gzip (1.10-4+deb11u1) over (1.10-4) ...
Setting up gzip (1.10-4+deb11u1) ...
(Reading database ... 35585 files and directories currently installed.)
[... 62 lines cut ...]
Preparing to unpack .../17-userconf-pi_0.2_all.deb ...
[... cut ...]
Setting up raspi-config (20220506) ...
Setting up libxml2:arm64 (2.9.10+dfsg-6.7+deb11u2) ...
Setting up userconf-pi (0.2) ...
Setting up libraspberrypi-bin (1:2+git20220324~090146+c4fd1b8-1) ...
Setting up dpkg-dev (1.20.10) ...
Setting up libraspberrypi-dev (1:2+git20220324~090146+c4fd1b8-1) ...
Setting up rpi-eeprom (13.14-1) ...
Setting up libraspberrypi-doc (1:2+git20220324~090146+c4fd1b8-1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u2) ...
Processing triggers for man-db (2.9.4-2) ...
```

Marshalldog, 2022-06-10:
```
pi@raspberrypi:~ $ curl -sSL https://raw.githubusercontent.com/TobKra96/music_le                                                                              d_strip_control/master/setup.sh | sudo bash -s -- -b master

                  *********************
                  *  Installing MLSC  *
                  *********************

[1/4] Updating and installing required packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3 is already the newest version (3.9.2-3).
python3 set to manually installed.
The following additional packages will be installed:
  git-man libasound2-dev libatlas3-base liberror-perl libgfortran5 libjack-dev
  libjack0 libportaudio2 libportaudiocpp0 python-pip-whl python3-distutils
  python3-lib2to3 python3-setuptools python3-wheel uuid-dev
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn libasound2-doc libatlas-doc
  liblapack-doc jackd1 portaudio19-doc python-setuptools-doc
Recommended packages:
  python3-dev
The following NEW packages will be installed:
  git git-man libasound2-dev libatlas-base-dev libatlas3-base liberror-perl
  libgfortran5 libjack-dev libjack0 libportaudio2 libportaudiocpp0
  portaudio19-dev python-pip-whl python3-distutils python3-lib2to3 python3-pip
  python3-setuptools python3-wheel uuid-dev
0 upgraded, 19 newly installed, 0 to remove and 0 not upgraded.
Need to get 16.0 MB of archives.
After this operation, 80.1 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 liberror-perl all 0.17029                                                                              -1 [31.0 kB]
Get:2 http://deb.debian.org/debian bullseye/main arm64 git-man all 1:2.30.2-1 [1                                                                              ,827 kB]
Get:3 http://deb.debian.org/debian bullseye/main arm64 git arm64 1:2.30.2-1 [5,4                                                                              28 kB]
Get:4 http://deb.debian.org/debian bullseye/main arm64 libgfortran5 arm64 10.2.1                                                                              -6 [333 kB]
Get:5 http://archive.raspberrypi.org/debian bullseye/main arm64 libasound2-dev a                                                                              rm64 1.2.4-1.1+rpt2 [126 kB]
Get:6 http://deb.debian.org/debian bullseye/main arm64 libatlas3-base arm64 3.10                                                                              .3-10 [2,278 kB]
Get:7 http://archive.raspberrypi.org/debian bullseye/main arm64 python-pip-whl a                                                                              ll 20.3.4-4+rpt1 [2,002 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main arm64 python3-pip all                                                                               20.3.4-4+rpt1 [337 kB]
Get:9 http://deb.debian.org/debian bullseye/main arm64 libatlas-base-dev arm64 3                                                                              .10.3-10 [2,419 kB]
Get:10 http://deb.debian.org/debian bullseye/main arm64 libjack0 arm64 1:0.125.0                                                                              -3+b1 [101 kB]
Get:11 http://deb.debian.org/debian bullseye/main arm64 uuid-dev arm64 2.36.1-8+                                                                              deb11u1 [99.6 kB]
Get:12 http://deb.debian.org/debian bullseye/main arm64 libjack-dev arm64 1:0.12                                                                              5.0-3+b1 [223 kB]
Get:13 http://deb.debian.org/debian bullseye/main arm64 libportaudio2 arm64 19.6                                                                              .0-1.1 [61.3 kB]
Get:14 http://deb.debian.org/debian bullseye/main arm64 libportaudiocpp0 arm64 1                                                                              9.6.0-1.1 [18.4 kB]
Get:15 http://deb.debian.org/debian bullseye/main arm64 portaudio19-dev arm64 19                                                                              .6.0-1.1 [105 kB]
Get:16 http://deb.debian.org/debian bullseye/main arm64 python3-lib2to3 all 3.9.                                                                              2-1 [77.8 kB]
Get:17 http://deb.debian.org/debian bullseye/main arm64 python3-distutils all 3.                                                                              9.2-1 [143 kB]
Get:18 http://deb.debian.org/debian bullseye/main arm64 python3-setuptools all 5                                                                              2.0.0-4 [366 kB]
Get:19 http://deb.debian.org/debian bullseye/main arm64 python3-wheel all 0.34.2                                                                              -1 [24.0 kB]
Fetched 16.0 MB in 12s (1,391 kB/s)
Selecting previously unselected package liberror-perl.
(Reading database ... 35602 files and directories currently installed.)
Preparing to unpack .../00-liberror-perl_0.17029-1_all.deb ...
Unpacking liberror-perl (0.17029-1) ...
Selecting previously unselected package git-man.
Preparing to unpack .../01-git-man_1%3a2.30.2-1_all.deb ...
Unpacking git-man (1:2.30.2-1) ...
Selecting previously unselected package git.
Preparing to unpack .../02-git_1%3a2.30.2-1_arm64.deb ...
Unpacking git (1:2.30.2-1) ...
[... cut ...]
Backup of existing /usr/share/alsa/alsa.conf created.

New configuration for /usr/share/alsa/alsa.conf saved.

[4/4] Creating autostart service for MLSC...

Autostart script for MLSC already exists in '/etc/systemd/system/mlsc.service'.

                  ********************************************
                  *       MLSC installation completed!       *
                  * Please reboot your system (sudo reboot). *
                  ********************************************
pi@raspberrypi:/share/music_led_strip_control/server $
```

Marshalldog, 2022-06-10:
i rebooted and manually install the libraries within : https://www.youtube.com/watch?v=ShpOVoOpqrQ&feature=youtu.be
but some of the libraries had errors
eg.
```
pi@raspberrypi:~ $ pip install PyAudio
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting PyAudio
  Using cached PyAudio-0.2.11.tar.gz (37 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: PyAudio
  Building wheel for PyAudio (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-3.9
      copying src/pyaudio.py -> build/lib.linux-aarch64-3.9
      running build_ext
      building '_portaudio' extension
      creating build/temp.linux-aarch64-3.9
      creating build/temp.linux-aarch64-3.9/src
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c src/_portaudiomodule.c -o build/temp.linux-aarch64-3.9/src/_portaudiomodule.o
      src/_portaudiomodule.c:28:10: fatal error: Python.h: No such file or directory
         28 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for PyAudio
  Running setup.py clean for PyAudio
Failed to build PyAudio
Installing collected packages: PyAudio
  Running setup.py install for PyAudio ... error
  error: subprocess-exited-with-error

  × Running setup.py install for PyAudio did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      running install
      running build
      running build_py
      creating build
      creating build/lib.linux-aarch64-3.9
      copying src/pyaudio.py -> build/lib.linux-aarch64-3.9
      running build_ext
      building '_portaudio' extension
      creating build/temp.linux-aarch64-3.9
      creating build/temp.linux-aarch64-3.9/src
      aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c src/_portaudiomodule.c -o build/temp.linux-aarch64-3.9/src/_portaudiomodule.o
      src/_portaudiomodule.c:28:10: fatal error: Python.h: No such file or directory
         28 | #include "Python.h"
            |          ^~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> PyAudio

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

Marshalldog, 2022-06-10:
tried doing a clean install with pi lite 64 bit 4-4-2022
got this
```
pi@raspberrypi:~ $ sudo apt-get update
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [44                                                                                                                                                             .1 kB]
Hit:2 http://deb.debian.org/debian bullseye InRelease
Get:3 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 Pa                                                                                                                                                             ckages [152 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main armhf Pa                                                                                                                                                             ckages [154 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main Translat                                                                                                                                                             ion-en [95.3 kB]
Get:7 http://deb.debian.org/debian bullseye-updates InRelease [39.4 kB]
Get:8 http://deb.debian.org/debian bullseye-updates/main armhf Packages [2,604 B                                                                                                                                                             ]
Get:9 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [2,600 B                                                                                                                                                             ]
Get:10 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [273 k                                                                                                                                                             B]
Get:11 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [279 k                                                                                                                                                             B]
Fetched 1,065 kB in 4s (281 kB/s)
Reading package lists... Done
pi@raspberrypi:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  cifs-utils dphys-swapfile dpkg dpkg-dev gzip libcamera0 libdpkg-perl
  libldap-2.4-2 libldap-common liblzma5 libpam-chksshpwd libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin
  libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libssl1.1 libxml2
  openssl pi-bluetooth raspi-config rpi-eeprom rsyslog userconf-pi xz-utils
28 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.9 MB of archives.
After this operation, 14.3 MB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:1 http://security.debian.org/debian-security bullseye-security/main arm64 dp                                                                                                                                                             kg arm64 1.20.10 [2,513 kB]
Get:2 http://security.debian.org/debian-security bullseye-security/main arm64 gz                                                                                                                                                             ip arm64 1.10-4+deb11u1 [129 kB]
Get:3 http://security.debian.org/debian-security bullseye-security/main arm64 rs                                                                                                                                                             yslog arm64 8.2102.0-2+deb11u1 [681 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main arm64 li                                                                                                                                                             blzma5 arm64 5.2.5-2.1~deb11u1 [165 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main arm64 xz                                                                                                                                                             -utils arm64 5.2.5-2.1~deb11u1 [219 kB]
Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 ci                                                                                                                                                             fs-utils arm64 2:6.11-3.1+deb11u1 [87.8 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main arm64 dp                                                                                                                                                             kg-dev all 1.20.10 [2,306 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main arm64 libpam0g arm64 1                                                                                                                                                             .4.0-9+deb11u1+rpt2 [128 kB]
Get:9 http://security.debian.org/debian-security bullseye-security/main arm64 li                                                                                                                                                              [... cut ...]
Autostart script for MLSC created in '/etc/systemd/system/mlsc.service'.
Do you want to enable autostart for MLSC? [y/N] Y
Proceeding...
Created symlink /etc/systemd/system/multi-user.target.wants/mlsc.service → /etc/systemd/system/mlsc.service.

Autostart for MLSC enabled.

                  ********************************************
                  *       MLSC installation completed!       *
                  * Please reboot your system (sudo reboot). *
                  ********************************************
```

Marshalldog, 2022-06-10:
```
pi@raspberrypi:~ $ sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-pip is already the newest version (20.3.4-4+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ pip3 install --upgrade pip
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/local/lib/python3.9/dist-packages (22.1.2)
pi@raspberrypi:~ $ sudo pip3 install -I numpy==1.17.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy==1.17.0
  Downloading numpy-1.17.0.zip (6.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 6.5/6.5 MB 7.9 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: numpy
  Building wheel for numpy (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [360 lines of output]
      Running from numpy source directory.
      blas_opt_info:
      blas_mkl_info:
      customize UnixCCompiler
        libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      blis_info:
      customize UnixCCompiler
        libraries blis not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      openblas_info:
      customize UnixCCompiler
      customize UnixCCompiler
        libraries openblas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      atlas_3_10_blas_threads_info:
      Setting PTATLAS=ATLAS
      customize UnixCCompiler
        libraries tatlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      atlas_3_10_blas_info:
      customize UnixCCompiler
        libraries satlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      atlas_blas_threads_info:
      Setting PTATLAS=ATLAS
      customize UnixCCompiler
        libraries ptf77blas,ptcblas,atlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
        NOT AVAILABLE

      atlas_blas_info:
      customize UnixCCompiler
      get_default_fcompiler: matching types: '['gnu95', 'intel', 'lahey', 'pg', 'absoft', 'nag', 'vast', 'compaq', 'intele', 'intelem', 'gnu', 'g95', 'pathf95', 'nagfor']'
      customize Gnu95FCompiler
      Could not locate executable gfortran
      Could not locate executable f95
      customize IntelFCompiler
      Could not locate executable ifort
      Could not locate executable ifc
      customize LaheyFCompiler
      Could not locate executable lf95
[... 682 lines cut ...]
          self.build_sources()
        File "/tmp/pip-install-kfmkzdhj/numpy_3ba2b7bf82da4b038ef7075179267dc1/numpy/distutils/command/build_src.py", line 159, in build_sources
          self.build_extension_sources(ext)
        File "/tmp/pip-install-kfmkzdhj/numpy_3ba2b7bf82da4b038ef7075179267dc1/numpy/distutils/command/build_src.py", line 316, in build_extension_sources
          sources = self.generate_sources(sources, ext)
        File "/tmp/pip-install-kfmkzdhj/numpy_3ba2b7bf82da4b038ef7075179267dc1/numpy/distutils/command/build_src.py", line 369, in generate_sources
          source = func(extension, build_dir)
        File "numpy/core/setup.py", line 430, in generate_config_h
          moredefs, ignored = cocache.check_types(config_cmd, ext, build_dir)
        File "numpy/core/setup.py", line 49, in check_types
          out = check_types(*a, **kw)
        File "numpy/core/setup.py", line 286, in check_types
          raise SystemError(
      SystemError: Cannot compile 'Python.h'. Perhaps you need to install python-dev|python-devel.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> numpy

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

Marshalldog, 2022-06-10:
```
 sudo pip3 install scipy==1.3.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting scipy==1.3.0
  Downloading scipy-1.3.0.tar.gz (23.6 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 23.6/23.6 MB 5.9 MB/s eta 0:00:00
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [781 lines of output]
      Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
      Ignoring numpy: markers 'python_version == "3.5"' don't match your environment
      Ignoring numpy: markers 'python_version == "3.6"' don't match your environment
      Collecting wheel
        Using cached https://www.piwheels.org/simple/wheel/wheel-0.37.1-py2.py3-none-any.whl (35 kB)
      Collecting setuptools
        Using cached https://www.piwheels.org/simple/setuptools/setuptools-62.3.3-py3-none-any.whl (1.2 MB)
      Collecting Cython>=0.29.2
        Using cached Cython-0.29.30-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.manylinux_2_24_aarch64.whl (1.8 MB)
      Collecting numpy==1.14.5
        Downloading numpy-1.14.5.zip (4.9 MB)
           ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.9/4.9 MB 8.0 MB/s eta 0:00:00
        Preparing metadata (setup.py): started
        Preparing metadata (setup.py): finished with status 'done'
      Building wheels for collected packages: numpy
        Building wheel for numpy (setup.py): started
        Building wheel for numpy (setup.py): finished with status 'error'
        error: subprocess-exited-with-error

        × python setup.py bdist_wheel did not run successfully.
        │ exit code: 1
        ╰─> [362 lines of output]
            Running from numpy source directory.
            /tmp/pip-install-l22gjf3y/numpy_fe79018f1ef849afb16fee3953d4075f/numpy/distutils/misc_util.py:464: SyntaxWarning: "is" with a literal. Did you mean "=="?
              return is_string(s) and ('*' in s or '?' is s)
            blas_opt_info:
            blas_mkl_info:
            customize UnixCCompiler
              libraries mkl_rt not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE

            blis_info:
            customize UnixCCompiler
              libraries blis not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE

            openblas_info:
            customize UnixCCompiler
            customize UnixCCompiler
              libraries openblas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE

            atlas_3_10_blas_threads_info:
            Setting PTATLAS=ATLAS
            customize UnixCCompiler
              libraries tatlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE

            atlas_3_10_blas_info:
            customize UnixCCompiler
              libraries satlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE

            atlas_blas_threads_info:
            Setting PTATLAS=ATLAS
            customize UnixCCompiler
              libraries ptf77blas,ptcblas,atlas not found in ['/usr/local/lib', '/usr/lib', '/usr/lib/aarch64-linux-gnu/atlas', '/usr/lib/aarch64-linux-gnu']
              NOT AVAILABLE
[... 708 lines cut ...]
                out = check_types(*a, **kw)
              File "numpy/core/setup.py", line 279, in check_types
                raise SystemError(
            SystemError: Cannot compile 'Python.h'. Perhaps you need to install python-dev|python-devel.
            [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: legacy-install-failure

      × Encountered error while trying to install package.
      ╰─> numpy

      note: This is an issue with the package mentioned above, not pip.
      hint: See above for output from the failure.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

TobKra96, 2022-06-11:
Hi,

it looks like, one or more libs need the build libs for python.
Please try to install: 

`sudo apt-get install python-dev`
`sudo apt-get install python3-dev`
`sudo apt-get install python-devel`
`sudo apt-get install python3-devel`

Marshalldog, 2022-06-13:
lol 32bit workds perfectly but 64 doesnt

closed 2022-06-13 by Marshalldog