TobKra96/music_led_strip_control #169: SK6812 RGBW Weise Led nicht einzel ansteuerbar

dano0894, 2022-06-15:
Hallo ich habe folgendes problem:
Von der Audiosteuerung funktioniert eigentlich alles aber wen ich im nonaudio bereich beim singel effekt einstellen will, dass er nur die einzelne weise LED ansteuert dann bleibt der streifen leider dunkel

Vielen Dank für den Suport
Tolles Projekt 😋 

Liebe Grüsse

TobKra96, 2022-10-09:
Hey, hast du geprüft, ob der richtige LED Streifen in den Device Settings ausgewählt ist? Ich habe gerade leider keinen RGBW Streifen zur Hand.

paddrix, 2023-09-10:
Hi, falls es noch relevant ist... hatte das selbe Problem und habe es wie folgt gefixt: `"SK6812"` durch `"sk6812"` ersetzen in `server/libs/outputs/output_raspi.py` und `server/libs/effects/effect_single.py`.

Bei meinem speziellen RGBW LED Stripe waren dann die Farben noch vertauscht, was ich durch folgende Zeilen in `libs/outputs/output_raspi.py` behoben habe:
```python
w = np.left_shift(output_array[3][:].astype(int), 24) + pylint: disalbe=assignment-from-no-return
g = np.left_shift(output_array[1][:].astype(int), 16) + pylint: disalbe=assignment-from-no-return
r = np.left_shift(output_array[0][:].astype(int), 8) + pylint: disalbe=assignment-from-no-return
b = output_array[2][:].astype(int)
grbw = np.bitwise_or(np.bitwise_or(np.bitwise_or(r, w), g), b).astype(np.uint32)
```

dano0894, 2023-09-13:
Werds mir anschauen…Herzlichen DankVon meinem iPhone gesendet

closed 2026-10-31 by TobKra96