TobKra96/music_led_strip_control #174: Can't install on 64bit RaspiOS Lite

Xitee1, 2022-07-06:
MLSC does not install on 64bit RaspiOS Lite. 32bit works without errors.

Both, 32bit and 64bit were completely fresh installed and I have only executed 'sudo su', 'apt update' and 'apt full-upgrade -y' before the installation.

```
root@MLSC:~# curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b master

                  *********************
                  *  Installing MLSC  *
                  *********************

[1/4] Updating and installing required packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-10).
git is already the newest version (1:2.30.2-1).
portaudio19-dev is already the newest version (19.6.0-1.1).
python3 is already the newest version (3.9.2-3).
python3-pip is already the newest version (20.3.4-4+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/local/lib/python3.9/dist-packages (22.1.2)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

Packages updated and installed.

[2/4] Installing MLSC...
MLSC is already installed. Do you want to reinstall it? [y/N] y
Proceeding...

New backup of MLSC created.
Cloning into 'music_led_strip_control'...
remote: Enumerating objects: 259, done.
remote: Counting objects: 100% (259/259), done.
remote: Compressing objects: 100% (222/222), done.
remote: Total 259 (delta 55), reused 137 (delta 16), pack-reused 0
Receiving objects: 100% (259/259), 2.55 MiB | 235.00 KiB/s, done.
Resolving deltas: 100% (55/55), done.

Config is stored in .mlsc, in the same directory as the MLSC installation.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting PyAudio==0.2.11
  Downloading PyAudio-0.2.11.tar.gz (37 kB)
  Preparing metadata (setup.py) ... done
Collecting numpy==1.21.0
  Downloading numpy-1.21.0-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (13.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.0/13.0 MB 307.0 kB/s eta 0:00:00
Collecting coloredlogs==15.0
  Downloading https://www.piwheels.org/simple/coloredlogs/coloredlogs-15.0-py2.py3-none-any.whl (45 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.8/45.8 kB 242.8 kB/s eta 0:00:00
Collecting scipy==1.6.2
  Downloading scipy-1.6.2-cp39-cp39-manylinux2014_aarch64.whl (26.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 26.1/26.1 MB 507.6 kB/s eta 0:00:00
Collecting Werkzeug==2.0.3
  Downloading https://www.piwheels.org/simple/werkzeug/Werkzeug-2.0.3-py3-none-any.whl (289 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 289.2/289.2 kB 538.2 kB/s eta 0:00:00
Collecting Flask==2.0.3
  Downloading https://www.piwheels.org/simple/flask/Flask-2.0.3-py3-none-any.whl (95 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 95.6/95.6 kB 638.8 kB/s eta 0:00:00
Collecting Flask_Login==0.5.0
  Downloading https://www.piwheels.org/simple/flask-login/Flask_Login-0.5.0-py2.py3-none-any.whl (16 kB)
Collecting Jinja2==3.0.1
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.0.1-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.7/133.7 kB 32.7 MB/s eta 0:00:00
Collecting waitress==2.1.2
  Downloading https://www.piwheels.org/simple/waitress/waitress-2.1.2-py3-none-any.whl (57 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 57.7/57.7 kB 596.9 kB/s eta 0:00:00
Collecting rpi_ws281x==4.3.4
  Downloading rpi_ws281x-4.3.4.tar.gz (64 kB)
[... 164 lines cut ...]
  Stored in directory: /tmp/pip-ephem-wheel-cache-jxml272k/wheels/87/fe/e6/fc8deeb581a41e462eafaf19fee96f51cdc8391e0be1c8088a
Successfully built pyrsistent
Failed to build PyAudio rpi_ws281x psutil
ERROR: Could not build wheels for psutil, which is required to install pyproject.toml-based projects

[3/4] Configuring microphone settings...

Backup of existing /etc/asound.conf created.

New configuration for /etc/asound.conf saved.

Backup of existing /usr/share/alsa/alsa.conf created.

New configuration for /usr/share/alsa/alsa.conf saved.

[4/4] Creating autostart service for MLSC...

Autostart script for MLSC already exists in '/etc/systemd/system/mlsc.service'.

                  ********************************************
                  *       MLSC installation completed!       *
                  * Please reboot your system (sudo reboot). *
                  ********************************************
root@MLSC:~#
```

Teraskull, 2022-07-06:
Hi @Xitee1,

Yes, it's a known issue, since we did not test or support Raspberry Pi OS 64bit yet.

Can you try running:
```shell
sudo apt-get install libglib2.0-dev python3-pyaudio python3-psutil
```

And then run the MLSC setup again?

This should install the dependencies required to build wheels for `PyAudio`, `rpi_ws281x`, and `psutil`.

If this will solve your issue, I will add this installation step to the setup.

Thanks!

Teraskull, 2022-07-06:
@Xitee1,

Everything installed correctly for me after running:
```shell
sudo apt install python3-dev
```

Teraskull, 2022-07-08:
Closing as fixed.

closed 2022-07-08 by Teraskull