TobKra96/music_led_strip_control #181: branch dev_2.3 doesn't work after PI upgrade to bullseye

fherbet, 2022-09-17:
Hi MLSC team !

tried recently to upgrade my PI3B+ to Bullseye after i noticed python3.7 was the standard py version in Buster and that branch dev_2.3 only works with py38+ ...

i just updated APT source list and ran "dist-upgrade"

obv, i lost networking and wifi config but that's ok i've managed to recover it by plugging a screen and using raspi-config.

i really need that branch for following reason :
https://github.com/TobKra96/music_led_strip_control/issues/138

master now works perfectly but branch dev_2.3 gives that screen :

[image]

i just did

```
sudo systemctl stop mlsc.service

pi@ledstrip:/share/music_led_strip_control $ sudo git checkout dev_2.3 
Basculement sur la branche 'dev_2.3'
Votre branche est à jour avec 'origin/dev_2.3'.

sudo systemctl start mlsc.service

sudo systemctl status mlsc.service

● mlsc.service - Music LED Strip Control
     Loaded: loaded (/etc/systemd/system/mlsc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-09-17 13:23:09 BST; 7s ago
   Main PID: 6496 (python3)
      Tasks: 29 (limit: 2058)
        CPU: 10.250s
     CGroup: /system.slice/mlsc.service
             ├─6496 python3 main.py
             ├─6503 python3 main.py
             ├─6504 python3 main.py
             ├─6505 python3 main.py
             ├─6506 python3 main.py
             ├─6511 python3 main.py
             ├─6512 python3 main.py
             ├─6513 python3 main.py
             ├─6514 python3 main.py
             ├─6515 python3 main.py
             └─6516 python3 main.py

sept. 17 13:23:14 ledstrip python3[6512]: INFO     - libs.effect_service            - Starting Effect Service component from device: STRIPLEDM>
sept. 17 13:23:14 ledstrip python3[6512]: INFO     - libs.effect_service            - Effects component started. Device: STRIPLEDMAIN
sept. 17 13:23:14 ledstrip python3[6503]: INFO     - libs.device                    - Starting device: STRIPLED1ST
sept. 17 13:23:14 ledstrip python3[6513]: INFO     - libs.output_service            - Starting Output service... Device: STRIPLED2ND
sept. 17 13:23:14 ledstrip python3[6514]: INFO     - libs.effect_service            - Starting Effect Service component from device: STRIPLED2>
sept. 17 13:23:14 ledstrip python3[6515]: INFO     - libs.output_service            - Starting Output service... Device: STRIPLED1ST
sept. 17 13:23:14 ledstrip python3[6514]: INFO     - libs.effect_service            - Effects component started. Device: STRIPLED2ND
sept. 17 13:23:14 ledstrip python3[6516]: INFO     - libs.effect_service            - Starting Effect Service component from device: STRIPLED1>
sept. 17 13:23:14 ledstrip python3[6516]: INFO     - libs.effect_service            - Effects component started. Device: STRIPLED1ST
sept. 17 13:23:15 ledstrip python3[6505]: INFO     - waitress                       - Serving on http://0.0.0.0:8080

```

when i try to refresh server page, i get 

```
pi@ledstrip:/share/music_led_strip_control $ sudo systemctl status mlsc.service
● mlsc.service - Music LED Strip Control
     Loaded: loaded (/etc/systemd/system/mlsc.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2022-09-17 17:02:17 BST; 6min ago
   Main PID: 3949 (python3)
      Tasks: 29 (limit: 2058)
        CPU: 11min 56.041s
     CGroup: /system.slice/mlsc.service
             ├─3949 python3 main.py
             ├─3959 python3 main.py
             ├─3960 python3 main.py
             ├─3961 python3 main.py
             ├─3962 python3 main.py
             ├─3967 python3 main.py
             ├─3968 python3 main.py
             ├─3969 python3 main.py
             ├─3970 python3 main.py
             ├─3971 python3 main.py
             └─3972 python3 main.py

sept. 17 17:08:26 ledstrip python3[3961]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
sept. 17 17:08:26 ledstrip python3[3961]:   File "/usr/local/lib/python3.9/dist-packages/flask_login/utils.py", line 290, in decorated_view
sept. 17 17:08:26 ledstrip python3[3961]:     return current_app.ensure_sync(func)(*args, **kwargs)
sept. 17 17:08:26 ledstrip python3[3961]:   File "/share/music_led_strip_control/server/libs/app/home/routes.py", line 17, in index
sept. 17 17:08:26 ledstrip python3[3961]:     groups = Executer.instance.device_executer.get_assigned_groups()
sept. 17 17:08:26 ledstrip python3[3961]:   File "/share/music_led_strip_control/server/libs/webserver/executer_base.py", line 24, in wrapper
sept. 17 17:08:26 ledstrip python3[3961]:     return func(*args, **kwargs)
sept. 17 17:08:26 ledstrip python3[3961]:   File "/share/music_led_strip_control/server/libs/webserver/blueprints/device_executer.py", line 113, in get_assigned_groups
sept. 17 17:08:26 ledstrip python3[3961]:     for group_id, group_name in global_groups.items():
sept. 17 17:08:26 ledstrip python3[3961]: AttributeError: 'list' object has no attribute 'items'

```

what should i check now ? (before restarting PI install from scratch ... btw what's the recommended PI OS version?)

Thanks for your help....!

\\ fherbet

Teraskull, 2022-09-17:
Hi @fherbet,

It's a known issue, branch `dev_2.3` has some ongoing WIP features, and has bugs.

If you want the latest stable commit of this branch, checkout to commit feb66ba5905c9340a5c61c9f9a2495d919875325:
```shell
git checkout feb66ba5905c9340a5c61c9f9a2495d919875325
```

Anything after the commits starting with [WIP] is not guaranteed to be stable.

Regarding the OS version, the latest 32bit or 64bit Pi OS is recommended.

Thanks!

fherbet, 2022-09-18:
Thanks for your quick support!

i confirm that this workaround did the trick...

as not being a git expert , i had to do this before checkout the right commit (fort those who might struggle like me!)
```
git pull origin feb66ba5905c9340a5c61c9f9a2495d919875325

git rebase --skip

git checkout feb66ba5905c9340a5c61c9f9a2495d919875325
```

love the new features especially edit link on effects + groups which i'll test via API a bit later

any ideas when dev_2.3 will merge into master?

again , no rush at all ! just out of curiosity... thanks for your wonderful project... will keep on adding a couple of nodeMCU client !
that rocks!

Teraskull, 2022-09-18:
Hey @fherbet!

I just pushed a fix for this bug, you can now test the latest commit as well, no need for the workaround.

Thanks for reporting! No date for the merge yet, as there is a big feature update for groups, and we need to make sure these types of bugs don't happen in `master`.

fherbet, 2022-09-18:
Cool! i just tested it and can confirm it works as expected. thanks very much.

i can test groups feature but would require some guidance... i understood that this was available through API correct ?
i see you can form groups through GUI but group effects have to be triggered via API ..correct ?

Teraskull, 2022-09-18:
You can read in the commit message for the [WIP] 3fa1c818df0be95c81f0b1da0179ca1576a4c0d8 about what works and what does not, but yes, all the group stuff can be used with the API, except for changing effect settings.

Forming a group is just adding the same tag to multiple devices, so it is possible through API of course.

The UI/UX for multiple selected effects on the dashboard is still in progress.

Have fun testing and hopefully, finding bugs :)

closed 2022-09-20 by Teraskull