TobKra96/music_led_strip_control #182: Normal (non individually addressable) LED strip

ex0nuss, 2022-10-04:
Hi there,

is it possible to use only a "simple" RGBW LED-strip without individually addressable LEDs? 
It would be really nice to get blinking / color chaining lights with a spare Raspberry Pi but without the need for more expensive individually addressable strips.

Thx in advance and regards!

Teraskull, 2022-10-04:
Hi,

In the current setup, no, it's not possible, because the GPIO connection expects the DIN data connection to the addressable LED strip.

We have no plans for supporting 2 different strip types as there are already other options available, either on Arduino, or a Pi.
Unless someone else is willing to support this feature.

Thank you for understanding.

ex0nuss, 2022-10-04:
Hi,

thank you for your quick reply!
I totally understand that it would take much effort implementing a different type of LED strip.

Do you have any recommendations for projects with single color LEDs?

TobKra96, 2022-10-09:
Hi, 

as far as I see, most Raspberry Pi Projects use the more advanced addressable LED Stripes.
There are some simple Arduino projects. Maybe this will help you?

ex0nuss, 2022-10-09:
I'll take a look, thanks!

closed 2022-10-09 by ex0nuss