TobKra96/music_led_strip_control #183: 2 led strips not working

EL3CTR0B1TCH, 2022-10-08:
Every time I connect 2 led strips everything works fine for a few mins but then one of the strips just freezes in a certain state and stops working no matter what I do in the gui. Both are configured right and both work independently, no inverter on each one of them and both are getting 5v straight from rpi4.

TobKra96, 2022-10-09:
Hi,

have you checked our FAQ? https://github.com/TobKra96/music_led_strip_control/wiki/FAQ#issues-with-the-led-strip-driven-by-the-raspberry-pi

If you want to drive 2 different stripes on the Rapberry Pi, you have to use 
- Different direct memory access channel (DMA)
- Select the PWM channel (0 or 1). 
- Dependent on the channel, you have to select the GPIO pin (Channel 0 - 12, 18 and Channel 1 - 13, 19 (only available on B+ and later))

You can configure them inside MLSC Webinterface on the device settings. 

Additionally, you have to disable the system audio, which uses one of the PWM modules that you need. See https://github.com/jgarff/rpi_ws281x#pwm

toreupfeet, 2022-10-23:
Rpi4 8gb here, and I also cannot add a second output without corruption. When both outputs are set to DMA 10, this is actually the best outcome. When I configure the two outputs to separate DMA (as instructed in the FAQ) there is extreme corruption in the data signal; very strange behaviors on _both_ strips. I'm using WS2812B. 

When there is only one "Device" with Rpi output, that device is stable, but as soon as I even click Create New Device (immediately after I click that button ...I haven't even saved the device yet) the LEDs in the first output become unstable, and they remain unstable no matter what I try.

toreupfeet, 2022-10-23:
By the way @EL3CTR0B1TCH, if I'm interpreting what you wrote correctly, you're driving the LED strips directly from your Pi4 GPIO pins, which are not 5V logic but actually 3.3V. If so you could be straining your Pi4. It's worth passing your data output through a shifter like a 74HC14. I've had luck with a 74HC14 in my brief trials, but others claim only an HCT will do the job. See here:

https://electronics.stackexchange.com/questions/473337/raspberry-pi-with-74hc14-connecting-to-led-strip-ws2812b

42Willow, 2023-07-02:
Any update? I have the same issue here.