TobKra96/music_led_strip_control #185: Version for PC

JinxQoida, 2022-10-18:
Hello,
I know this might  sound crazy but wouldn`t it be possible to port  MLSC  to be available on PC? 
This is just to  use the internal  line in/mic ports on the build in  sounds card.
I take an assumption of little machine ready with wifi card for   managing MLSC  nodes. only without abilit of its own   RGB led control. 
I am in total disregard of power consumption here, this is not an issue for me.

Teraskull, 2022-10-19:
Hi, unfortunately, it's not really possible to run MLSC on Windows, because of a lot of python modules being only available for Linux and the way multiprocessing is done on Linux.

JinxQoida, 2022-10-19:
I was more toward  having a linux (ubuntu or any other)  on this little PC not Windows. 
I tried already installing but there are some dependencies that force the MLSC application to  stop if they are not met.

ahsu-2019, 2022-11-26:
any update on this? looking for pi alternatives since pi 4s are in short supply.

bentstan, 2023-02-23:
I just tried to run it on my Laptop with Ubuntu installed besides windows (dual boot).
automatic setup script didn't work fully, there were some libs which were missing afterwards. Just trying to run the server and install all needed libs manually with pip did indeed work out for me.
This could also just be needed because of wrong linked python/pip version.

Just tested it with an external 3.5mm microphone. LED Device is a WLED Client in the same local network on port 19446 as mentioned in another issue here. Did work! 
Now, I just need to figure out how to set up a loopback audio device to use the system audio as input.

ACyberCat, 2023-04-08:
I just noticed the issue is regarding a native app for PC platform to eliminate the use of an RPi, but for those looking for "integration" with PC or any other platform for that matter, my reply is:

I made a simple windows app that uses the REST API that is provided to send commands to MLSC. 

While i do not think it is easy, there must be a way to emulate an audio device from within the device running MLSC that listens to incoming audio streams so you can route audio through the network to it and use that for microphone input or for native music playback.

Chase295, 2023-04-16:
Hello

I also want to install it on a ubunto server version. Do you have any tips for me. When I try to install scipy or something else this error comes 

Error: metadata-generation-failed

TobKra96, 2023-06-06:
Hi,
at the moment we only recommend using the Raspberry Pi OS 32 Bit.
Other OS could have compatibility issues with some of the libs as you have seen.
If you want to use another OS you have to solve the compatibility issues on your own. Sorry.

closed 2023-06-06 by TobKra96