TobKra96/music_led_strip_control #188: [dev_2.3] config_adjuster misbehaves if `old_config` doesn't have any `device_groups` keys

marcelgerber, 2023-01-04:
Hey and thank you so much for this very cool project!
I just updated from 2.2 stable to `dev_2.3`, and in doing so got the following error:
```
Jan 04 01:14:20 raspberrypi systemd[1]: Started Music LED Strip Control.
Jan 04 01:14:26 raspberrypi python3[501]: INFO     - __main__                       - Initializing MLSC...
Jan 04 01:14:26 raspberrypi python3[501]: INFO     - libs.config_converter.config_converter_v3 - Upgrade config to vers>
Jan 04 01:14:26 raspberrypi python3[501]: Traceback (most recent call last):
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/main.py", line 177, in <module>
Jan 04 01:14:26 raspberrypi python3[501]:     main.start()
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/main.py", line 82, in start
Jan 04 01:14:26 raspberrypi python3[501]:     self._config_instance.check_compatibility()
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/libs/config_service.py", line 1>
Jan 04 01:14:26 raspberrypi python3[501]:     loaded_config = config_converter_service.upgrade(loaded_config)
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/libs/config_converter/config_co>
Jan 04 01:14:26 raspberrypi python3[501]:     old_config = config_converter_v3.upgrade(old_config)
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/libs/config_converter/config_co>
Jan 04 01:14:26 raspberrypi python3[501]:     upgrade_step_2 = self.format_device_groups(upgrade_step_1)
Jan 04 01:14:26 raspberrypi python3[501]:   File "/share/music_led_strip_control/server/libs/config_converter/config_co>
Jan 04 01:14:26 raspberrypi python3[501]:     device_groups = old_config["device_configs"][device]["device_groups"]
Jan 04 01:14:26 raspberrypi python3[501]: KeyError: 'device_groups'
Jan 04 01:14:26 raspberrypi systemd[1]: mlsc.service: Main process exited, code=exited, status=1/FAILURE
Jan 04 01:14:26 raspberrypi systemd[1]: mlsc.service: Failed with result 'exit-code'.
Jan 04 01:14:26 raspberrypi systemd[1]: mlsc.service: Consumed 2.527s CPU time.
```

Your [code there assumes](https://github.com/TobKra96/music_led_strip_control/blob/013c5dd07781d50caac6592c04afbd6d0ad88340/server/libs/config_converter/config_converter_v3.py#L31) that keys like `general_settings.device_groups` already exist; however they don't exist if MSLC has never run off some earlier version of `dev_2.3` before, since the old config format only got introduced [in this commit](https://github.com/TobKra96/music_led_strip_control/commit/3fa1c818df0be95c81f0b1da0179ca1576a4c0d8) to begin with.

Teraskull, 2023-01-04:
Hi, thanks for the helpful bug feedback!

As I am completely refactoring the API, I will be able to fix this after I'm done, to properly test everything.

For now you can just manually add `"device_groups": {}` to your config file inside `general_settings` and also for each device inside `device_configs.YOUR_DEVICE`.

marcelgerber, 2023-01-04:
Yeah, sure! I could easily fix it on my end, just didn't want it to go unnoticed when 2.3 is published.

closed 2023-08-27 by Teraskull