TobKra96/music_led_strip_control #193: Scipy 1.6.2 dosn´t is not existent

Chase295, 2023-04-16:
Hello, 

i would like to install MLSC but when i copy paste the atomatik instalation link this issue will come. When i will install it manuel came the same issue.

Sry for my bad english i am from germany
[image]

Teraskull, 2023-04-17:
Hi,

Master branch is outdated and not supported. Please install the dev_2.3 branch using:
```shell
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3
```

BjoernRave, 2023-05-06:
When trying to start the server on the `2.3` branch I get: 
```
INFO     - __main__                       - Initializing MLSC...
INFO     - libs.config_converter.config_converter_v3 - Upgrade config to version 3.
Traceback (most recent call last):
  File "/share/music_led_strip_control/server/main.py", line 177, in <module>
    main.start()
  File "/share/music_led_strip_control/server/main.py", line 82, in start
    self._config_instance.check_compatibility()
  File "/share/music_led_strip_control/server/libs/config_service.py", line 140, in check_compatibility
    loaded_config = config_converter_service.upgrade(loaded_config)
  File "/share/music_led_strip_control/server/libs/config_converter/config_converter_service.py", line 23, in upgrade
    old_config = config_converter_v3.upgrade(old_config)
  File "/share/music_led_strip_control/server/libs/config_converter/config_converter_v3.py", line 19, in upgrade
    upgrade_step_1 = self.format_global_groups(old_config)
  File "/share/music_led_strip_control/server/libs/config_converter/config_converter_v3.py", line 31, in format_global_groups
    global_groups = old_config["general_settings"]["device_groups"]
KeyError: 'device_groups'
```

TobKra96, 2023-06-06:
Hi, looks like a compatibility issue with the configuration. Please delete the configuration file while the mlsc is stopped.
The config is located "../../.mlsc/"

closed 2023-06-06 by TobKra96