TobKra96/music_led_strip_control #195: Master branch doesnt work - IndexError: list index out of range in init_audio_service:

BjoernRave, 2023-05-16:
When trying to start the application after running the automatic script I get the following error:

```
INFO     - __main__                       - Initializing MLSC...
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi:1
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM hdmi:1
INFO     - __main__                       - Initialization finished.
INFO     - __main__                       - MLSC started...
INFO     - libs.audio_process_service     - Found the following audio sources:
ERROR    - libs.audio_process_service     - ********************************************************
ERROR    - libs.audio_process_service     - *                      Error                           *
ERROR    - libs.audio_process_service     - ********************************************************
ERROR    - libs.audio_process_service     - Could not find the mic with the id: 0
ERROR    - libs.audio_process_service     - Using the first mic as fallback.
ERROR    - libs.audio_process_service     - Please change the id of the mic inside the config.
ERROR    - libs.audio_process_service     - Could not init AudioService.
ERROR    - libs.audio_process_service     - Unexpected error in init_audio_service: list index out of range
Traceback (most recent call last):
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 86, in init_audio_service
    self.selected_device = self._devices[0]
IndexError: list index out of range
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/share/music_led_strip_control/server/libs/webserver/webserver.py", line 83, in start
    serve(self.server, host='0.0.0.0', port=webserver_port, threads=8)
  File "/usr/local/lib/python3.9/dist-packages/waitress/__init__.py", line 13, in serve
    server = _server(app, **kw)
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 78, in create_server
    last_serv = TcpWSGIServer(
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 244, in __init__
    self.bind_server_socket()
  File "/usr/local/lib/python3.9/dist-packages/waitress/server.py", line 361, in bind_server_socket
    self.bind(sockaddr)
  File "/usr/local/lib/python3.9/dist-packages/waitress/wasyncore.py", line 396, in bind
    return self.socket.bind(addr)
OSError: [Errno 98] Address already in use
```

TobKra96, 2023-06-06:
Hi, this sounds like the mlsc is already running in the background or the audio interface has issues.

at the moment we do not support the 64 bit OS. We haven't tested it. Maybe the pyaudio lib has some troubles.

sameert123, 2023-10-09:
Hi, 

I'm also getting a very similar issue using RPI 4B w Tonor G11 - . Have tried troubleshootinng on my own but no dice. If anyone found a solution or has an update it would be much appreciated! 

Thank you!

"INFO     - __main__                       - Initialization finished.
INFO     - __main__                       - MLSC started...
INFO     - libs.audio_process_service     - Found the following audio sources:
ERROR    - libs.audio_process_service     - ********************************************************
ERROR    - libs.audio_process_service     - *                      Error                           *
ERROR    - libs.audio_process_service     - ********************************************************
ERROR    - libs.audio_process_service     - Could not find the mic with the id: 0
ERROR    - libs.audio_process_service     - Using the first mic as fallback.
ERROR    - libs.audio_process_service     - Please change the id of the mic inside the config.
ERROR    - libs.audio_process_service     - Could not init AudioService.
ERROR    - libs.audio_process_service     - Unexpected error in init_audio_service: list index out of range
Traceback (most recent call last):
  File "/share/music_led_strip_control/server/libs/audio_process_service.py", line 86, in init_audio_service
    self.selected_device = self._devices[0]"

Teraskull, 2023-10-09:
@sameert123 Can you please try the `dev_2.3` branch? The `master` branch does not have many fixes that are in `dev_2.3`.

sameert123, 2023-10-09:
> @sameert123 Can you please try the `dev_2.3` branch? The `master` branch does not have many fixes that are in `dev_2.3`.

Hey - sorry total newb here how do I switch the branch? I just used the curl automated installation from the setup guide

Teraskull, 2023-10-09:
@sameert123 Use this command:

```bash
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3
```

sameert123, 2023-10-10:
> @sameert123 Use this command:
> 
> ```shell
> curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3
> ```

Hey - when i try a fresh dev2.3 install i get this error 

Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [43 lines of output]
      + meson setup /tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388 /tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388/.mesonpy-16dyfoz6/build -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388/.mesonpy-16dyfoz6/build/meson-python-native-file.ini
      The Meson build system
      Version: 1.2.2
      Source dir: /tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388
      Build dir: /tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388/.mesonpy-16dyfoz6/build
      Build type: native build
      Project name: SciPy
      Project version: 1.11.1
      C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C linker for the host machine: cc ld.bfd 2.35.2
      C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C++ linker for the host machine: c++ ld.bfd 2.35.2
      Cython compiler for the host machine: cython (cython 0.29.36)
      Host machine cpu family: arm
      Host machine cpu: arm
      Program python found: YES (/usr/bin/python3)
      Found pkg-config: /usr/bin/pkg-config (0.29.2)
      Run-time dependency python found: YES 3.9
      Program cython found: YES (/tmp/pip-build-env-h6kg0bo0/overlay/bin/cython)
      Compiler for C supports arguments -Wno-unused-but-set-variable: YES
      Compiler for C supports arguments -Wno-unused-function: YES
      Compiler for C supports arguments -Wno-conversion: YES
      Compiler for C supports arguments -Wno-misleading-indentation: YES
      Library m found: YES
      
      ../../meson.build:82:0: ERROR: Unknown compiler(s): [['gfortran'], ['flang'], ['nvfortran'], ['pgfortran'], ['ifort'], ['ifx'], ['g95']]
      The following exception(s) were encountered:
      Running `gfortran --version` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `gfortran -V` gave "[Errno 2] No such file or directory: 'gfortran'"
      Running `flang --version` gave "[Errno 2] No such file or directory: 'flang'"
      Running `flang -V` gave "[Errno 2] No such file or directory: 'flang'"
      Running `nvfortran --version` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `nvfortran -V` gave "[Errno 2] No such file or directory: 'nvfortran'"
      Running `pgfortran --version` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `pgfortran -V` gave "[Errno 2] No such file or directory: 'pgfortran'"
      Running `ifort --version` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifort -V` gave "[Errno 2] No such file or directory: 'ifort'"
      Running `ifx --version` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `ifx -V` gave "[Errno 2] No such file or directory: 'ifx'"
      Running `g95 --version` gave "[Errno 2] No such file or directory: 'g95'"
      Running `g95 -V` gave "[Errno 2] No such file or directory: 'g95'"
      
      A full log can be found at /tmp/pip-install-4ce4ap7f/scipy_b9435c2638ee486da6d6ea8fac748388/.mesonpy-16dyfoz6/build/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

no clue what this means - any advice?

Teraskull, 2023-10-11:
@sameert123 Looks like some package can't be installed on your pi. What's your Pi OS version?

sameert123, 2023-10-11:
It’s the latest 32bit Rpi OS Lite running on a 4B+

sameert123, 2023-10-11:
Should I retry w the full OS

Teraskull, 2023-10-11:
Try installing a 64bit OS