TobKra96/music_led_strip_control #204: automatic Installation not working BUG

makeitready, 2024-05-02:
Hi there,
i get a failture while installing the MLSC software with the automatic installation. It sais **"no module named coloredlogs"**
i already googled it but a fix lik "python -m pip install coloredlogs" does not help.

Pleas tell me what to do, tahnks a lot!

Putty Image:
[image]

#bug #install

Teraskull, 2024-05-02:
Hi, thanks for the error log!

That does look strange and I will be testing the installer to see if I can replicate this.

Are you trying to install the `master` branch or `dev_2.3`? If `master`, please try reinstalling the newer branch:

```sh
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3
```

If the issue still happens, I guess there were some breaking changes from `pip` that enforce PEP 668 and now Raspberry Pi OS prevents mixing packages from `apt` and `pip`.

One solution would be for me to use a virtual environment in the installer. Hopefully this won't break a lot of things further down the line.

Thanks again! I'll get to this as soon as possible!

carlyd95, 2024-05-03:
Oh, dang I just posted a similar issue recently. Glad to see this.

So far the only thing I found to work is `pip3 install -r req.txt --break-system-packages` with req.txt being all the packages in requirements.txt without specified versions.

I am going to try the new branch that you mentioned above!

Thanks! @Teraskull

carlyd95, 2024-05-03:
Using `curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/master/setup.sh | sudo bash -s -- -b dev_2.3` still yields the following

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

Teraskull, 2024-05-03:
Yea, it's preventing mixing `apt` and `pip` packages. I will test installing packages to a virtual environment to see if that works.

carlyd95, 2024-05-03:
@Teraskull I tried that with virtualenv but was unsuccessful. I hope you are able to find a work-around. :)

makeitready, 2024-05-03:
I flashed the SD Card of my Rpi4 completely and installed Raspberry Pi OS again.
This time i intstalled the dev_2.3 branch instead of the master branch as @Teraskull suggested. 

But this time a get a Error **"No module named pyaudio"**
Can you tell me what to do? I would like to use it next week 😆 

Putty:
[image]

thanks!

Teraskull, 2024-05-03:
@carlyd95 @makeitready I noticed that Debian 12 Bookworm distros are now forcing installing `pip` packages into a `venv`, but I'm currently trying to figure out the root access for `pip`, since MLSC requires root access to the microphone device.

Until I test and update the setup script, I doubt that even manually creating a `venv` will help, since the installer expects to use the global python package location, which is not possible anymore with Debian 12.

Hopefully this will require minimum changes to the installation, but I'm glad that  using `venv` is forced now, because previously the python/pip ecosystem managing was a mess.

## Workaround while I fix the installer on the latest distro:
For now, your best option is to install the legacy [Pi OS 11 Bullseye](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy-64-bit) if you want to run MLSC as soon as possible, since that distro was tested and the installation script is confirmed to be working there. But I will try to make it work on the latest distro in the following days.

Thanks for your patience!

Teraskull, 2024-05-06:
@carlyd95 @makeitready ,

Can you please try installing MLSC with this command?
```shell
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/dev_2.3/setup.sh | sudo bash -s -- -b dev_2.3
```

Just pushed the script update to the `dev_2.3` branch. The `master` branch won't work anyways because some Python modules there are too outdated for Python 3.11, so if all goes well, I will just merge the `dev_2.3` branch into `master` and bump the versions in `requirements.txt` after more testing.

I did not test the script on a Pi OS 12, since I don't have a spare SD card at the moment, but it did work on an x64 Debian 12 VM, so assuming nothing else regarding the process changed, it should work. Including installing pre-built wheels for some modules.

I recommend deleting the whole `/share` directory so that the script does a fresh install:
```shell
sudo rm -r /share
```

Let me know if you have any issues, thanks!

makeitready, 2024-05-22:
Hey there,

Your work-arround with the Bullseye OS works, thanks!

I also tried a new Install with Pi OS 64 Bit. I flashed the SD card and installed it with the command you posted above. Sadly, i have the same Errors like in my last post. 
Error **"No module named pyaudio"**

[image]

Thanks for all your work.

> @carlyd95 @makeitready ,
>
> Can you please try installing MLSC with this command?
>
> ```shell
> curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/dev_2.3/setup.sh | sudo bash -s [...]

Teraskull, 2024-05-22:
Hi @makeitready ,

Can you try running it like this, inside `/share/music_led_strip_control/server`?
```shell
sudo /share/venv/bin/python3 main.py
```

Since now a virtual environment is used, the Python executable should be from that venv.

You can call it like `/share/venv/bin/python3`, or you can temporarily make the venv Python the main one, by activating the virtual environment:
```shell
source /share/venv/bin/activate
```

makeitready, 2024-05-22:
Hi @Teraskull 

The installation with a RPI 4 and PI OS 64 Bit works with 
curl -sSL https://raw.githubusercontent.com/TobKra96/music_led_strip_control/dev_2.3/setup.sh | sudo bash -s -- -b dev_2.3
and
sudo /share/venv/bin/python3 main.py

But no the service dhcpcd does not work. Error "dhcp.serice not found"

[image]

So the WiFi Network does not appear...

Teraskull, 2024-05-22:
Hey @makeitready ,

Turns out Debian Bookworm replaced `dhcpcd` with [`NetworkManager`](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/#:~:text=rather%20than%20PulseAudio.-,Networking,-Graphics%20and%20sound), and there is no easy way to get `dhcpcd` running on it without breaking stuff.

They also replaced `Alsa` with [`PipeWire`](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/#:~:text=in%20due%20course.-,PipeWire,-So%2C%20we%E2%80%99ve%20changed), so that change will break the music effects and microphone detection too.

I guess, for now the best option is to use Debian Bullseye, because there are too many breaking changes in the new OS. 

Unfortunately, I do not have time or the possibility to test Bookworm on a Pi now, and I have no idea how to migrate the network and microphone configurations to support NetworkManager/PipeWire. I think a lot of software will have problems with the Bullseye -> Bookworm upgrade too.

Thanks for your issue reports and support!

closed 2026-10-31 by TobKra96